SpenceKonde/ReleaseScripts #3: Replace wetware release scripting

SpenceKonde, 2015-09-26:
Currently, updating the json file is handled manually by per1234. 

Ideally, this should be implemented in a bash script that can run on drazzy.com along with the other scripts.

MCUdude, 2020-05-09:
I'm not sure if it is of any help since your boards manager packages are different than mine, but I've created a script that pretty much automates everything. All I need to do is to create a release and run the script. It downloads the latest release, extracts it, removes the `avr` folder, compresses it, and adds a new entry to the json file. All I have to do then is to push the new tar.bz2 file and the json file. You can find the script here:
https://github.com/MCUdude/MiniCore/blob/gh-pages/Boards_manager_release.sh

SpenceKonde, 2021-02-27:
Thanks, obcviously this is what I am using now.

closed 2021-02-27 by SpenceKonde