SpenceKonde/ReleaseScripts #5: Entry for newly added URL doesn't appear the first time Boards Manager is opened

per1234, 2015-11-03:
See https://github.com/arduino/Arduino/issues/3795 for an explanation of the issue. Can you confirm this issue? Arduino IDE 1.6.6 was released today with this issue(doesn't happen in 1.6.5) so if this is real(it happens consistently for me) I think a note should be added to the Boards Manager installation instructions for both cores, otherwise users are going to think the Boards Manager URL they added didn't work when the entry doesn't show up.

SpenceKonde, 2015-11-03:
I'll test tonight... goddamnit...

SpenceKonde, 2015-11-04:
Confirmed. I've updated the readme's appropriately. Is there anywhere else I should update it?

per1234, 2015-11-04:
If you would add a comment to the issue confirming that it also happens for you it might help get priority on fixing it. I hope they will put out a 1.6.6r2 soon.

closed 2015-11-05 by per1234