SpenceKonde/ReleaseScripts #35: megaTinyCore 2.2.1 package size is incorrect in

stonehippo, 2021-01-10:
In `packaga_drazzy.com_index.json`, the current package size is listed as 741852 for version 2.2.1 of megaTinyCore (the same as 2.2.0). Based on the file downloaded, the correct size is 742369.

This mismatch in size causes arduino-cli to fail when installing or upgrading the package, with the following error:

```
Error during install: installing platform megaTinyCore:megaavr@2.2.1: checking local archive integrity
```

Correcting the size of the package locally fixes the error.

SpenceKonde, 2021-01-10:
Thanks. Huh. Was I looking at the wrong file? I remember being like "what?! How could it be the same size" and re-checking it a couple of times...

Will be fixed in 2.2.2 - another serious issue has been reported with 2.2.1 on linux platforms.

stonehippo, 2021-01-10:
Yeah, I don’t know :)

When the CLI barfed, I when and dug into it’s code to see how it checks the file, figuring maybe the SHA-256 was off. When I noticed 2.2.0 and 2.2.1 where the same size, I went and checked the downloaded .tar.xz. Go figure.

Anyhow, I’ll stand by for 2.2.2.

stonehippo, 2021-01-18:
Fixed with 2.2.2+

closed 2021-01-18 by stonehippo