SpenceKonde/ReleaseScripts #36: Transitioning to new hostname

SpenceKonde, 2021-03-03:
I am in the process of general rebranding to Azduino - and I think this is a @per1234 question

The json file should move to Azduino.com as well. (among other things, I have https set up there which several people have requested for the json file). But I am wondering how I should handle the json file - what, if anything, should get changed?
Should it be renamed to package_azduino.com_index.json? I recall reading that the file was "supposed" to be named package_domaintname_index.json, so that would seem to be preferable? 

Or is the file name set in stone for the cores, and only new releases under the new json file could change that? 

What else should I know?

per1234, 2021-03-03:
>Should it be renamed to package_azduino.com_index.json? I recall reading that the file was "supposed" to be named package_domaintname_index.json, so that would seem to be preferable?

The reason for the recommendation of using the domain name in a specific format in the package index file name is that every package index file is stored together in the data directory, so if any two boards package authors pick the same file name then those two won't be able to be used at the same time. Every domain name is unique, so if everyone followed this recommendation faithfully then there would be no chance of name collisions.

But this is only a recommendation. There is no mechanism to enforce it. https://azduino.com/package_foobar_index.json will be accepted just as well as https://azduino.com/package_azduino.com_index.json.

>Or is the file name set in stone for the cores, and only new releases under the new json file could change that?

The tricky thing is about changing the URL. The URL is changing whether the file name stays the same or not. The old URL is in people's **File > Preferences > Additional Boards Manager URLs** setting, and the equivalent sort of thing in automation scripts. So everyone must change their URL. If the old URL becomes no longer accessible then ATTinyCore will disappear from the **Tools > Board** menu the first time the Arduino IDE tries to download the package index file from the old URL and fails (it  deletes the existing package index file before attempting to download the new one, so if the download fails then the package index file is lost).

Fortunately, Arduino IDE 2.x and Arduino CLI are able to handle this situation correctly. I just did some tests and wasn't able to find any way to cause problems with a dead package index URL or a leftover outdated package_drazzy.com_index.json alongside an updated package_azduino.com_index.json. However, it's likely the large majority of the users will be on the old Arduino IDE at least until the production release of Arduino IDE 2.x comes out.

But if you will still have drazzy.com for some time and you don't mind maintaining two copies of the package index, then you could make a soft transition where the documentation recommends the new Boards Manager URL, but the old URL continues to work and provide updates. To encourage people to update from the old URL, you could use that trick we did for the "ATtiny Modern" platform where a deprecation notice was added to the platform name:
```
      "platforms": [
        {
          "name": "ATtiny Modern(deprecated, use ATTinyCore instead)",
```

SpenceKonde, 2021-03-03:
Okay! So, it's no problem to change to package_azduino.com_index.json. 

Great. 
As for the URL - I will retain drazzy.com indefinitely.... my main question would be how does it interact with server-side redirects? ie, can I just put a redirect up there and have it work seamlessly?