SpenceKonde/ReleaseScripts #38: Arduino board manager throws an error when preferences refer to ATtinyCore

softfoot, 2022-01-21:
I recently installed v1.8.19 and have begun to get the attached error whenever I use the library or board manager.

It seems to be caused by adding "http://drazzy.com/package_drazzy.com_index.json" to arduino's preferences page.

It sometimes also shows a "non-trusted contribution" warning and ignors a script.

Is there a workaround for this ?? as I am working on a project for the AtTiny.

[image]

Regards,
Dave

per1234, 2022-01-21:
>have begun to get the attached error

There is some previous discussion about the new "`Index error: could not find referenced tool name=avrdude version=6.3.0-arduino17or18 packager=arduino`" error in my non-viable proposal here: https://github.com/SpenceKonde/ReleaseScripts/pull/37

>It sometimes also shows a "non-trusted contribution" warning and ignors a script.

The "`Warning: non trusted contribution, skipping script execution`" warning is unrelated to the new error. See the discussion here for the explanation of why it is the intended behavior: https://github.com/SpenceKonde/ATTinyCore/issues/569

SpenceKonde, 2022-01-22:
Marking as critical because I am getting constant bug reports about this. it is causing concern among all users of all three cores. Is there any way to fix this?!

per1234, 2022-01-23:
I have investigated another potential solution, which is removing the `megaTinyCore:megaavr@2.5.8` entry from the package index:

```diff
diff --git a/package_drazzy.com_index.json b/package_drazzy.com_index.json
index 8898e38..adba395 100644
--- a/package_drazzy.com_index.json
+++ b/package_drazzy.com_index.json
@@ -5438,49 +5438,6 @@
             }
           ]
         },
-        {
-          "name": "megaTinyCore",
-          "architecture": "megaavr",
-          "version": "2.5.8",
-          "category": "Contributed",
-          "url": "https://SpenceKonde.github.io/megaTinyCore/megaTinyCore-2.5.8.tar.bz2",
-          "archiveFileName": "megaTinyCore-2.5.8.tar.bz2",
-          "checksum": "SHA-256:6fcf791f2083588f628950a3fb6f810f98f5b445e4c54c75d0549adfe33c9669",
-          "size": "690530",
-          "boards": [
-            {
-              "name": "Full Arduino support for the tinyAVR 0-series, 1-series, and the new 2-series!</br>\n        24-pin parts: ATtiny3227/3217/1627/1617/1607/827/817/807/427<br/>\n        20-pin parts: ATtiny3226/3216/1626/1616/1606/826/816/806/426/416/406<br/>\n        14-pin parts: ATtiny3224/1624/1614/1604/824/814/804/424/414/404/214/204<br/>\n        8-pin parts: ATtiny412/402/212/202<br/>"
-            },
-            {
-              "name": "Optiboot serial bootloader supported (and optional) on all parts. <br/>"
-            },
-            {
-              "name": "2.5.8: Json file was botched, this version is no good. Use 2.5.9 2.5.7: This has a more graceful fix to the Wire buffer bug fixed in 2.5.5, fixes a bug reported with SerialUPDI which prevented standalone read (as opposed to verify) from working. This also fixed verbose mode and added tools -> programmer options for people who have problems running at anything but the slowest speed. Unlike previous 2.5.x, these bugs are not considered critical.<br/> 2.5.0: A HUGE update! New Wire implementation with master+slave support, improvements to serial including half-duplex support, a bunch of new features ported from DxCore and more! Be sure to check the documentation for the full details!"
-            }
-          ],
-          "toolsDependencies": [
-            {
-              "packager": "DxCore",
-              "name": "avr-gcc",
-              "version": "7.3.0-atmel3.6.1-azduino4b"
-            },
-            {
-              "packager": "arduino",
-              "name": "avrdude",
-              "version": "6.3.0-arduino17or18"
-            },
-            {
-              "packager": "arduino",
-              "name": "arduinoOTA",
-              "version": "1.3.0"
-            },
-            {
-              "packager": "megaTinyCore",
-              "version": "3.7.2-post1",
-              "name": "python3"
-            }
-          ]
-        },
         {
           "name": "megaTinyCore",
           "architecture": "megaavr",
```

---

Unfortunately, this solution also is not without impacts on the users.

## Arduino IDE 1.8.10 - 1.8.19

Under the following conditions:

- User is using Arduino IDE version in the range 1.8.10 - 1.8.19
- User has `megaTinyCore:megaavr@2.5.8` installed.

[... 34 lines cut ...]
With these Arduino IDE versions, having multiple installations of a given platform causes compilation to fail with a "`Board ___ (platform ___, package ___) is unknown`" error.

For example:

```
Board atxy7 (platform megaavr, package megaTinyCore) is unknown
```

This can only be fixed by manually deleting the folder containing the `megaTinyCore:megaavr@2.5.8` installation

## Arduino IDE 2.0.0 and Arduino CLI

Arduino CLI and Arduino IDE 2.x refused to install `megaTinyCore:megaavr@2.5.8`. This means that these users will not be affected in any way by the removal of `megaTinyCore:megaavr@2.5.8` from the package index.

Even if they installed `megaTinyCore:megaavr@2.5.8` via Arduino IDE 1.x, it will continue to work in Arduino CLI and Arduino IDE 2.x just like always and their Boards Manager will correctly clean up `megaTinyCore:megaavr@2.5.8` when another version is installed.

## Conclusion

I would say that the long term benefits of removing the warning outweigh the inconvenience caused to Arduino IDE 1.8.10-1.8.19 users.

Users of Arduino IDE 1.8.9 and older would probably prefer the warning to their megaTinyCore being broken. Hopefully most people have updated to a more modern version of the Arduino IDE. There might be some who are unable to due to the constraints of their computer (e.g., the last version of the IDE that can be used with Windows XP is 1.8.9).

One thing to consider is that the longer you wait to implement an impactful fix to the package index, the fewer people will be affected. The reason is that everyone will gradually upgrade from 2.5.8 to 2.5.9 and it is only those who have 2.5.8 installed who are impacted. The users who were quick to install 2.5.8 are likely also the same users who will be quick to update to 2.5.9. The slow updaters are likely still on <=2.5.7.

I'm happy to submit a PR for the proposed change if you like @SpenceKonde.

SpenceKonde, 2022-01-23:
Cool, thank you for the investigation.

(And, I reckon people are upgrading from 2.5.x to 2.5.9 pretty quickly - the intermediate versions were hosed pretty badly. And apparently there's still a critical bug that I was unaware of reported in 2.5.9....)

I think I'll drop 2.5.8 from the list shortly after I do 2.5.10, which is apparently going to have to be sooner than I'd hoped.... (by the reasoning of waiting longer = fewer people impacted). I have been telling people to use 1.8.13 or later since it came out, and started doing things that make the the UX miserable for people with older versions with reckless abandon and wanton disregard for their happiness over a year ago (I recently checked the date when I stopped working around the old programmer submenu behavior, because someone was still on an ancient version and complained). megaTinyCore and DxCore have got to be absolute hell without that change :-P )

SpenceKonde, 2022-01-25:
Make sure that you do not have 2.5.8 of megaTinyCore installed by midweek if you are using 1.8.9 of the IDE or earlier. The message can be ignored if you do not have megaTinyCore 2.5.8 installed (other versions, including 2.5.9, are fine), and the consequences are  not problematic no matter what version you have installed  if you are using a later version of the IDE. 

I will release 2.5.10 in the next couple of days (I hope to address #624; #623 is already serious enough that a fix needs to go out this week, even if this issue didn't exist) and make the version causing the error message disappear shortly thereafter, and hopefully there won't be many people still on 1.8.9 or earlier with but with 2.5.8 of megaTinyCore installed by the time I do it. 

(IMO, versions of the IDE prior to 1.8.13 should not be used anymore anyway - that update greatly improved the tools -> board and tools -> programmer menu when lots of cores are installed)

Erik84750, 2022-01-26:
I do not have megaTinyCore installed, but the ATTinyCore (1.5.2) is installed. IDE: 1.8.13.
When http://drazzy.com/package_drazzy.com_index.json is added to the Additional Board Manager then the Index error appears.
How can the solution proposed by per1234 be implemented?

SpenceKonde, 2022-02-20:
This was fixed several weeks ago.

closed 2022-02-20 by SpenceKonde