SpenceKonde/ReleaseScripts #40: XML failure in package_drazzy.com_index.json

uweng, 2022-09-12:
line **3241** in **file package_drazzy.com_index.json** is :

**},**

it must be just:

**}**

In arduino ide parsing the json file produces this failure:

Skipping contributed index file /home/user/.arduino15/package_drazzy.com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Unexpected character (']' (code 93)): expected a value
 at [Source: (FileInputStream); line: 3242, column: 12]
 at [Source: (FileInputStream); line: 3242, column: 11] (through reference chain: cc.arduino.contributions.packages.ContributionsIndex["packages"]->java.util.ArrayList[1]->cc.arduino.contributions.packages.ContributedPackage["tools"]->java.util.ArrayList[6]->cc.arduino.contributions.packages.ContributedTool["systems"])

stonehippo, 2022-09-13:
@uweng I'm pretty sure this is addressed by [e4de67f](https://github.com/SpenceKonde/ReleaseScripts/commit/e4de67fcbc89788dcd3800b9ce0b91956371059d). I confirmed the issue prior to updating the .json file and the fix after.

SpenceKonde, 2023-09-28:
Yeah you were doing that as I was doing a release and had slipped while updating the json. I think it was only like that for ten minutes or so.

closed 2023-09-28 by SpenceKonde