SpenceKonde/ReleaseScripts #41: HELP!!!

SpenceKonde, 2022-12-21:
Why isn't Build_DxCore working anymore??
```
jq: error: syntax error, unexpected IDENT, expecting '}' (Unix shell quoting issues?) at <top-level>, line 27:
      "name": "Supported UPDI programmers: SerialUPDI (serial adapter w/diode or resistor), jtag2updi, nEDBG, mEDBG, EDBG, SNAP, AtmelICE and PICkit4 or use one of those to load Optiboot (included) for serial programming if you determine that it is appropriate for your application."
jq: error: syntax error, unexpected IDENT, expecting ';' or ')' (Unix shell quoting issues?) at <top-level>, line 27:
      "name": "Supported UPDI programmers: SerialUPDI (serial adapter w/diode or resistor), jtag2updi, nEDBG, mEDBG, EDBG, SNAP, AtmelICE and PICkit4 or use one of those to load Optiboot (included) for serial programming if you determine that it is appropriate for your application."
jq: error: syntax error, unexpected and, expecting '}' (Unix shell quoting issues?) at <top-level>, line 27:
      "name": "Supported UPDI programmers: SerialUPDI (serial adapter w/diode or resistor), jtag2updi, nEDBG, mEDBG, EDBG, SNAP, AtmelICE and PICkit4 or use one of those to load Optiboot (included) for serial programming if you determine that it is appropriate for your application."                                                                             
jq: error: Invalid numeric literal at EOF at line 1, column 5 (while parsing '7.3.0') at <top-level>, line 34:
      "version": "7.3.0-atmel3.6.1-azduino6"
jq: error: Invalid numeric literal at EOF at line 1, column 4 (while parsing '.6.1') at <top-level>, line 34:
      "version": "7.3.0-atmel3.6.1-azduino6"
jq: error: Invalid numeric literal at EOF at line 1, column 5 (while parsing '6.3.0') at <top-level>, line 39:
      "version": "6.3.0-arduino17or18"
jq: error: Invalid numeric literal at EOF at line 1, column 5 (while parsing '1.3.0') at <top-level>, line 44:
      "version": "1.3.0"
jq: error: Invalid numeric literal at EOF at line 1, column 5 (while parsing '3.7.2') at <top-level>, line 48:
      "version": "3.7.2-post1",
jq: 8 compile errors
```

Floessie, 2022-12-22:
@SpenceKonde The solution is super simple: Just finalize the string [here]{https://github.com/SpenceKonde/ReleaseScripts/blob/master/build_DxCore.sh#L96} by adding a `"` in that line (and removing the empty one, and adding a `.` to the end of the sentence).

HTH,
Flössie

closed 2022-12-27 by SpenceKonde

Floessie, 2023-01-02:
De rien! :wink: