SpenceKonde/ReleaseScripts #42: http://drazzy.com/package_drazzy.com_index.json is not working anymore for Github actions

ArminJo, 2023-09-03:
Hi,
since a few days the URL http://drazzy.com/package_drazzy.com_index.json
is not working anymore, see https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/6041049256/job/16393168027#step:3:70.
Here, https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/5931549426/job/16083624590#step:3:70 it worked one week before.

Maybe Github force **https** now and has problems with the self signed certificate, as I have in Firefox with https://drazzy.com/package_drazzy.com_index.json.

Thanks for your work and for caring.
Best regards
Armin

SpenceKonde, 2023-09-03:
Transfering to tnhe repo concerned with board manager json and issuwa nor depenswenr on core

ArminJo, 2023-09-03:
Hi SpenceConde,
as you can see here: https://github.com/Arduino-IRremote/Arduino-IRremote/actions/runs/6041049256/workflow#L187
the compile action uses your working **http** URL.
My suspicion is, that **github switched internally to https only**.
So maybe if you **delete your https URL**, github is forced to use the http URL as specified in my script?
Thanks
Armin

SpenceKonde, 2023-09-28:
This is NOT related to SSL, it is https://github.com/arduino/arduino-cli/issues/2345

You can make the CI work by specifying a version of the CLI that predates the regression like this - I've done that to all my CI systems.
https://github.com/SpenceKonde/DxCore/blob/9f1b6385ca868f6c8c763a708028da419c76c8b4/.github/workflows/compile-examples.yml#L327

closed 2023-09-28 by SpenceKonde

ArminJo, 2023-09-28:
Thanks a lot for your research 🥇 
The 0.33.0 solved my problem!

SpenceKonde, 2023-10-15:
Also, finally, https works!